JaapBraam/LoRaWanGateway #2: time format

athtest800, 2016-07-06:
Hi again!
I use loraserver  [the](https://github.com/brocaar/loraserver) and i have the following error
`time="2016-07-06T10:50:57+03:00" level=info msg="gateway: received udp packet from gateway" addr=192.168.200.160:15274 type=PushData
time="2016-07-06T10:50:57+03:00" level=error msg="gateway: could not handle packet: parsing time \"\"2016-07-06T07:51:59.396440Z\"\" as \"\"2006-01-02 15:04:05 MST\"\": cannot parse \"T07:51:59.396440Z\"\" as \" \"" addr=192.168.200.160:15274 data_base64="Ad/FABj+NP//of9jeyJzdGF0Ijp7InJ4ZnciOjIsImFsdGkiOjAsImxhdGkiOjAuMCwicnhvayI6MiwiZGVzYyI6IkVTUDgyNjYgR2F0ZXdheSAoTHVhKSIsImxvbmciOjAuMCwicGZybSI6IkVTUDgyNjYiLCJ0aW1lIjoiMjAxNi0wNy0wNlQwNzo1MTo1OS4zOTY0NDBaIiwidHhuYiI6MCwibWFpbCI6IiIsImR3bmIiOjAsInJ4bmIiOjIsImFja3IiOjg4Ljh9fQ=="
time="2016-07-06T10:50:57+03:00" level=info msg="gateway: received udp packet from gateway" addr=192.168.200.160:15274 type=PullData
`

It seems that the format that the gateway sends it is wrong.

JaapBraam, 2016-07-07:
I try to follow https://raw.githubusercontent.com/Lora-net/packet_forwarder/master/PROTOCOL.TXT as precise as possible.

It states: 
time | string | UTC time of pkt RX, us precision, ISO 8601 'compact' format

I use The Things Network routers to test, and those routers understand the timeformat..

JaapBraam, 2016-09-21:
I fixed the date/time format in the stat message to be YYYY-MM-DD HH-MM-SS GMT.

closed 2016-09-21 by JaapBraam