JaapBraam/LoRaWanGateway #5: Different freq detected 902.2999xx instead of 9023xx(channel 0, subband 0)

shmrymbd, 2016-08-20:
My node are using LMIC code and transmitting to ESP gateway.. Setup US-915 channel sub band 0, 0-7 and enable channel 0 only. The freq received is different compare what I got if using Raspi Single packet gateway source. Below are the rcd packet;

rxpk.017f1000600194ffff0b94cb.message.{"rxpk":[{"rssi":-105,"stat":1,"modu":"LORA","rfch":0,"tmst":61998373,"datr":"SF10BW125","lsnr":-2,"time":"2016-08-20T02:28:08.167523Z","codr":"4/5","data":"QMSAzwYAVAEOGm77NG9CjwcPdOph","freq":902.299987,"chan":0,"size":21}]}.length.238

JaapBraam, 2016-08-22:
The frequency is converted from an integer value to a floating point frequency in Mhz. The lora chip cannot be configured listen on _exactly_ 902.3Mhz. The things network servers will round it up to 902.3Mhz...

closed 2016-08-22 by JaapBraam

shmrymbd, 2016-08-29:
I am using loraserver. It seem the packet rejected by the loraserver https://github.com/brocaar/loraserver

kragupat, 2016-11-08:
@shmrymbd 
Looks like you have this gateway running in US 915Mhz, 
Is all that is required to do this is change the local CHN array defnintion to 

local CHN={}
CHN[0]=chan(902300000,"LoRa",MC1.BW125)
CHN[1]=chan(902500000,"LoRa",MC1.BW125)
CHN[2]=chan(902700000,"LoRa",MC1.BW125)
CHN[3]=chan(902900000,"LoRa",MC1.BW125)
CHN[4]=chan(903100000,"LoRa",MC1.BW125)
CHN[5]=chan(903300000,"LoRa",MC1.BW125)
CHN[6]=chan(903500000,"LoRa",MC1.BW125)
CHN[7]=chan(903700000,"LoRa",MC1.BW125)
CHN[8]=chan(903900000,"LoRa",MC1.BW250)
CHN[9]=chan(904100000,"FSK" ,MC1.BW150)

or do I have to do something more.