JaapBraam/LoRaWanGateway #6: I can not run your code on my board,plz help

yuguangyu, 2016-08-29:
I am not familiar with nodemcu and I just followed your steps. 

I encountered a problems:
PANIC: unprotected error in call to Lua API (bad argument #1 to '?' (number expected, got nil))

and the nodemcu restart over and over. and I am confused.

JaapBraam, 2016-08-29:
Do you have the hardware settings in init.lua? It looks like the pin assignments for DIO0 and DIO1 can't be found.

in init.lua you need to set the pins you connected the DIO0 and DIO1 lines of your SX1276 to:
```
-- hardware setup
GW_DIO0=1
GW_DIO1=2
```

yuguangyu, 2016-08-30:
I got the source code from github and did not change them. and I will check the wire connection.

yuguangyu, 2016-08-30:
thanks a lot! I find the problem, the luaLoader I used uploaded the .lua wrongly, and I upload the .lua with ESPlorer again, and it runs quiete well.

closed 2016-08-30 by JaapBraam