JaapBraam/LoRaWanGateway #7: Slave Select pin

zoutepopcorn, 2016-09-12:
Hi,

Thanks for the amazing gateway. I did everything, but my gateway is not running. Tried to run
dofile("init.lua")

On Esplorer I uploaded:
init.lua
LoRaWanGW.lc 
SX1276.lc 

And then it dont print anything. Tried to include some prints at the end and beginning.

I'm quit new to nodemcu. I dont get errors only:
ntp synced using 89.xx.20.xx

Also have my Slave select on pin D8, so I changed in SX1276.lua:
gpiowrite(8, 0)
Maybe we can I can put it in init.lua, so we can choose the SS pin... 

Any solutions to debug??

JaapBraam, 2016-09-12:
Did you actually upload LoRaWanGW.lc and SX1276.lc? Normally LoRaWanGW.lua and SX1276.lua should be uploaded. The script in init.lua will compile them after a restart and throw away the lua files...

If you see 'ntp synced using ...' in the console, your connection to internet should be fine.
Next line you should see is 'router ip: ....' indicating the LoraWAN router that you are connected to.

I think that your changes regarding Slave Select are ok: if they were wrong you would see 'Unknown radio: ' in the console.

Next you should see only received packages in the console (with a line 'ntp synced using ... ' every five minutes.

It looks like you can communicate with the sx1276 chip, but you don't receive any messages.
Things to check:
   Are there really messages to receive on channel 0?
   Are the pins DIO0 and DIO1 connected correctly and configured in init.lua?

zoutepopcorn, 2016-09-12:
Thanks for quick reply. I will double check pin connection. I'm sure transmitting something on channel 0 ( i can receive with the arduino gateway from @platenspeler)

Then i will put some print() in to find out whats happening... 

I uploaden only .lua files :), and they compile fine with your init script.

JaapBraam, 2016-09-15:
Did you get it working?

zoutepopcorn, 2016-09-16:
Nope. Got the message Unknown radio now.... Will check it later with another board. 

In Lua is pin D8 -> 8 right?. Sometimes confusing the esp pinout...

JaapBraam, 2016-09-16:
The mapping between the nodemcu pin numbers and the ESP's gpio's can be found on http://nodemcu.readthedocs.io/en/dev/en/modules/gpio/

JaapBraam, 2016-09-21:
I have added the parameter GW_NSS to init.lua. This parameter defines the NSS pin to use.

closed 2016-09-21 by JaapBraam