JaapBraam/LoRaWanGateway #8: any suggestions to improve the performance ?

yuguangyu, 2016-09-21:
hi,
  Thanks for your single channel lorawan gateway, especially the ideal of CAD with scanning SF. It cause my arouse my interest. but I find problems as follow:
  The packet loss rate is pretty high, if CadDetected but receive packet failed,  the time wasted will influence the following SF scanning  procedure. It will cause packet loss. I do not know why The SF12 packet can cause the other SF CadDetected ? and the transmit payload also cause CadDetected.
    rx timeout    7    0    rssi    50
    rx timeout    8    0    rssi    45
  Looking forward to your suggestion. Thanks a lot

JaapBraam, 2016-09-21:
At the moment, a CAD or RX will be aborted when there is no signal anymore (RSSI below threshold). This is tested only once after the CAD or RX has started.

I've been looking for a good way to scedule a continuous detection of signal loss. I think I've tried everything possible to periodically ( about every 250us) test the RSSI from within the LUA runtime.

The idea is that when signal loss is detected the gateway can hop to the next channel an start a detection there. When hopping can be done fast enough it should be possible to support more than a single channel.... 

I've had some success receiving messages on more than one channel with an arduino/rfm95 combination, but it looked like the arduino was too slow (16Mhz clock); an esp8266 running on 160Mhz might do the trick.

At the moment I'm trying to develop an extra module in the nodemcu firware that can use a hardware timer to continuously check for signal loss. (see the dev branch in my nodemcu-firmware fork, app/modules/sx1276).

yuguangyu, 2016-09-22:
Your trial of supporting more than one channel by frequency hopping is quite creative, I hop that you can achieve good result.

My idea is :

- if single channel & multi SF listening on sx1278/1276 is reliable enough, the sensor node can obtain 1/3  probability to join success through single channel gateway. it can join success after several attempts , the result is acceptable.

- Then we can use ADR command to limit data transmit channel and SF of the node sensor. Through this method, A standard LoRaWAN node can communicate with single channel gateway with a longer join procedure . 

So, I want to improve the receive success rate with sx1278 configured one frequency and multi SF. I am looking for some ways to reduce wrong *CADDetected* trigger and Rx timeout. The effect of using RSSI  threshold  turns out not good.

 Communicating with you opens my mind and your idea of frequency hopping gives me a pleasant surprise. Thank you.

JaapBraam, 2016-10-05:
I discovered that one of the main problems is that the sx1276 gives high RSSI values when a signal is present on an adjacent channel. This causes the code to start the SF fallback mechanism on a channel, while the signal is not realy present on that channel. So the main goal at the moment is to get the sx1276 to only measure RSSI on the current channel. The thing I am investigating is whether the RSSI measurements can be more selective in FSK mode.

To be continued...

JaapBraam, 2017-01-28:
Multiple channels is a no-go for this setup

closed 2017-01-28 by JaapBraam

3dot3, 2017-02-01:
Maybe ESP32?

JaapBraam, 2017-02-01:
I ordered an ESP32 a while ago, but never got it delivered...

But I think that supporting multiple channels is not a performance or memory problem. The main issue is that the sx1276 doesn't seem to have a good enough channel separation to implement channel hopping. I succesfully received a message on channel 0 (868.1Mhz) that was transmitted on channel 1 (868.3Mhz)...

This makes the RSSI measurements unreliable: the signal on a channel can also be a stronger signal on an adjacent channel, causing the message to be missed.

cssiberica, 2017-03-15:
Maybe, two transceivers ?

We develop a complete board with ESP32 and rfm95w Whitecatboard ESP32N1 and working on these things, using SDR# and dongle on a pc, we are looking and adjusting some delays, maybe two transceivers will be good idea, or reverse engineering of spi bus on a SX1301 complete gateway board, like one from IMST, maybe some commands are missing on the datasheets :)

Also i think at the moment it's more important channel than spread factor, at the end with the down cost of the gateways if we got multichannel, can make each gateway for each SF, or leave nodes at same SF for example intermediate one SF9BW125.

Cheers,
Mike