JaapBraam/LoRaWanGateway #11: Component Question

forrestfiller, 2016-11-24:
Hi Jaap,

I'm getting started trying out your project, and I have a (I'm located in USA, so I use a 915MHz version)  [Adafruit RFM95W LoRa Radio Transceiver ](https://www.adafruit.com/products/3072).  Is that ok? Do you recommend any particular unit?

I was unable to figure out which SemTech radio Adafruit is using for the above product.

Thanks for your help!

Best regards,

Forrest

JaapBraam, 2016-11-24:
Hi Forrest,

The Adafruit page states that the SX1276 radio is used. This is exactly the one you need for this gateway.

I am using the bare RFM95W board ( without the Adafruit addons ). I expect your board to work just fine. As far as I can see the Adafruit board contains extra components to allow connection to 5v MCU's. The ESP8266 is 3.3v, so those extra components are not necessary.

In order to use the gateway in the USA you have to change the frequencies in the SX1276.lua file to the 915Mhz variants (line 75).

Regards,

Jaap

closed 2016-11-24 by JaapBraam

forrestfiller, 2016-11-24:
Thanks Jaap, sorry I missed that. Appreciate your time very much.