JaapBraam/LoRaWanGateway #13: Getting it to work with the Hallard WeMos board

dlarue, 2017-01-17:
I have a bunch of LoRa Single Channel Gateway by Hallard and would love to get full spread factor support.  But the Hallard board and WeMos have DIO0 and DIO1 OR'ed through diodes to one pin on the WeMos.  Would it still work?

JaapBraam, 2017-01-17:
I also use Hallard's Lora Single Channel Gateway board (the wemos d1 mini version).

I tried to make the full spread factor support working while using the DIO0 and DIO1 OR'ed to one pin, but it turned out that the strict timing requirements could not be met. The OR construction makes it necessary to first determine if it was DIO0 or DIO1 that triggered the interrupt. This takes to much time in LUA making the detection proces unreliable.

What I did is to not use the OR construction with diodes, but to connect DIO0 and DIO1 to seperate pins using small wires. I soldered wires from the diode mounting points to the SDA/SDL pins (D1/D2) on the gateway board, and configured the gateway to use D1/D2 for DIO0/DIO1

dlarue, 2017-01-17:
ok, I'll look into isolating diode-D2 so it can use D8 it's wired to and jumper diode-D3 to D3 since it's only used for LED output(not needed). This saves I2C(D1 and D2) if needed. Thanks, I'll give it a try today.

dlarue, 2017-01-17:
For some reason, WeMos pins D3 nor D4 would work, they are unused LED and Button pins on the Hallard pcb. Using D1 is working.  Tested a node on SF7 showed my 2 different gateways, a node on SF10 only showed going through the Lua gateway.  I think we can close this unless you have some insight into why pins D3 and D4 didn't work.

JaapBraam, 2017-01-17:
I think WeMos pins D3 and D4 also control the booting process. I didn't get it working using those pins either...

closed 2017-01-17 by JaapBraam