JaapBraam/LoRaWanGateway #15: OTAA worked!

dlarue, 2017-01-30:
I still have one of your gateways running(one of 4 here) and after having built a full multi-channel gateway I was doing some testing and fired up a node which I'd previously provisioned with OTAA code. I saw some strange messages on TTN data page and then realized it was the single channel node with OTAA I had setup long ago without any luck getting it to work.  Low and behold your gateway picked it up on the same Freq and up on TTN it showed data.

Fantastic!

Thanks for the great testbed and this might be an option for attendees of my workshops as long as they expect to use their nodes close to their homes or offices. IIRC you said the range of this multi SF gateway was down to a few hundred meters.

JaapBraam, 2017-01-30:
Your welcome!

closed 2017-01-30 by JaapBraam

webfrank, 2017-03-11:
Hi, could you provide a sample code for arduino+lmic working with oats and lua? I tried several configurations but at maximum I got 
recv	UNKNOWN	5555aaaae400000000000190010001575a502d49425431352d53000000000000000000000000000000000000000000003034356330363035633563610000000000000000000000000000000000000000373939393930313938000000000000000000000000000000000000000000000056352e302e302e313200000056352e300000000000000000000000000000000001000000000000375000401f282301003139322e3136382e312e313137000000000000003139322e3136382e312e310000000000000000003235352e3235352e3235352e300000000000000030343a35633a30363a30353a63353a6361000000

from lua console.

dlarue, 2017-03-11:
This is the standard IBM LMIC 1.5 example for OTAA with a couple of tweaks for Single Channel Operation(SCN).  It's been a while but I believe it did work. Granted, I used US915 Freq so changes required to make that happen were done.

`
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in config.h.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8]={ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }; //lsb
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }; //lsb
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
// The key shown here is the semtech default key.
static const u1_t PROGMEM APPKEY[16] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world! ";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60; //sec

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 0,
    .dio = {4, 5, 7},
};
[... 117 lines cut ...]
    // disable all but the first 8 channels
    //for (int channel=8; channel<8; ++channel) {
    //    LMIC_disableChannel(channel);
    //}
    //for (int channel=8; channel<72; ++channel) {
    //    LMIC_disableChannel(channel);
    //}
    Serial.println(F("Disabled some Channels"));

    // Maximum TX power
    LMIC.txpow = 27;
    Serial.println(F("TxPOW=27"));

    // expand the RX2 receive window by 1%  - Arduino inaccurate clock fix
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
    
    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}

`