JaapBraam/LoRaWanGateway #17: PANIC: unprotected error in call to Lua API (routing problem)

webfrank, 2017-03-01:
PANIC: unprotected error in call to Lua API (routing problem)

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 26844, room 16 
tail 12
chksum 0x12
ho 0 tail 12 room 4
load 0x3ffe8000, len 2192, room 12 
tail 4
chksum 0x5f
load 0x3ffe8890, len 136, room 4 
tail 4
chksum 0x5a
csum 0x5a
����o��r��o|�llld`��r�p��l���

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 5feae3fee1debe6aef75b8836fa86f9967037f2f
	SSL: false
	modules: bit,cjson,encoder,file,gpio,net,node,rtctime,sntp,spi,tmr,uart,wifi
 build 	built on: 2017-03-01 07:31
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
no config found, using default values
> got ip	192.168.1.123	255.255.255.0	192.168.1.1
Gateway ID	5CCF7FF42F2CD6BE
ntp synced using 213.239.154.12
2017-03-01 19:25:36 GMT
start allSF detector
router ip:	52.169.76.203
ntp synced using 94.228.220.14
rx timeout	7	rssi	50

JaapBraam, 2017-03-01:
This looks OK...

When the gateway is started for the first time, it will restart several times to compile all sources. 

When does your gateway restart with a 'routing problem' ? Just after boot or later on?

webfrank, 2017-03-01:
Hi, after few minutes of received packets it reboots with that error.

JaapBraam, 2017-03-01:
I have never experienced that myself. My two gateways are running for weeks without reboots...

I see you use a freshly built nodemcu firmware, I use the firmware that's in the firmware directory. Maybe the newer version has introduced an instability.

You can try to flash the firmware from the firmware directory and check if the problem still occurs.

webfrank, 2017-03-02:
Ok, I'll try

webfrank, 2017-03-02:
Problem is with otaa, with abp it works

JaapBraam, 2017-03-02:
The difference between otaa and abp from the gateway's perspective is that when using abp the gateway doesn't have to transmit messages to nodes. Maybe the problem occurs when the gateway has to transmit a message. When you send a downlink to your abp node, the gateway has to transmit that message.

Can you send data to your node using downlink on TTN's console?

JaapBraam, 2017-03-02:
The (routing problem) in the PANIC you report is triggered by the Espressif firmware (ESPCONN_RTE).

It looks like your ESP8266 is having network problems, check your wifi router...

closed 2017-03-09 by JaapBraam