JaapBraam/LoRaWanGateway #18: No data received on TTN

webfrank, 2017-03-01:
Hi, I'm receiving data from my test node:

rxpk	01f0db005ccf7ff42f2cd6be	message	{"rxpk":[{"rssi":-42,"stat":1,"modu":"LORA","rfch":1,"tmst":14490442,"datr":"SF7BW125","lsnr":8,"time":"2017-03-01T19:40:00.616803Z","codr":"4/5","data":"QNwSASYAAAABJPwj/fGgDeYDZRmgqqGoJbQ=","freq":868.100,"chan":0,"size":26}]}	length	240

but nothing appears on TTN console.

JaapBraam, 2017-03-01:
On https://staging.thethingsnetwork.org/gatewaystatus/ you can see that according to TTN your gateway status is  `5CCF7FF42FC23F28 active   active    now`.
That looks OK!

Are you sure your node and gateway are correctly registered via the TTN console? (console in bridge mode, correct keys for your node etc)?

webfrank, 2017-03-01:
Hi, I was using lsb netskey, now it works. Thanks.

closed 2017-03-01 by JaapBraam