JaapBraam/LoRaWanGateway #24: Operating on 433MHz

mogyoros, 2017-04-07:
Dear @JaapBraam ,

I am trying your code on an ESP12E, so far so good, nodemcu firmware installed, files uploaded.
Since I have an SX1278 module (only low band ie. 433), I've modified the freq setting to 433175000.
Connections:
DIO0-ESP12E GPIO5 - Definition in code - 1
DIO1-ESP12E GPIO4 - Definition in code - 2
SCK - ESP12E GPIO14
MISO - ESP12E GPIO12
MOSI - ESP12E GPIO13
NSS-ESP12E GPIO16 - Definition in code - 0
Gateway registers with backend, and can see the regular stats messages.

But I don't see packets arriving from the node (RN2483, also configured for 433 band) to the gateway. Is this something that should work, or am I trying something that is not implemented? - yet :-)
Thanks. Cheers.
Tamás

JaapBraam, 2017-04-07:
I've never tried.

The sx1276 I use should be programmed just like the sx1278, it is the same datasheet...

Do you see the text "Unknown radio:' in the logging? If you don't, communicating with your SX1278 is working correctly. It is a check of the chip type, and according to the datasheet it should return 0x12 for SX1276/77/78/79.

I would expect that it should just work...

JaapBraam, 2017-04-07:
I just did a quick test: I configured a gateway to listen on  433.175Mhz and hacked a node to use 433.175Mhz for channel 0 too.

It seems to work for my SX1276 based gateway... Even with 868Mhz antennas on my gateway and node...

`rxpk    018d260018fe34f42fde9937        message {"rxpk":[{"rssi":-118,"stat":1,"modu":"LORA","rfch":1,"tmst":715774550,"datr":"SF7BW125","lsnr":-5,"time":"2017-04-07T20:29:58.580898Z","codr":"4/5","data":"AIMBAPB+1bNwP7ZyC5VaCgBv7b90PJM=","freq":433.175,"chan":0,"size":23}]}     length  239
`
Are you sure your node is sending messages? Do you see anything in the log (rx timeouts etc)?

JaapBraam, 2017-04-07:
I've looked into it: TTN does not support 433Mhz (yet). 

I guess you have to use another network, or run your own, to use a 433Mhz gateway and nodes.

mogyoros, 2017-04-08:
Hi @JaapBraam ,

Thanks for your reply.

Glad you quickly tested yourself, that software wise it should work. I was on the same impression.

When I misconfigured the NSS setting in the Config.lua, yes I saw the Unrecognized transceiver = 255 message, but after finding out the correct naming convention, this has disappeared.

You are right, I am not using it on TTN, but on the openlora backend, which supports the 433 MHz frequency band networks also. But at this stage this is not relevant, since first I should see the above rxpk packets  on the gateway, which would state radio wise I am ok.

In the gateway I see timeouts, but nothing more:

`rx timeout    7       rssi    57`
`rx timeout      9       rssi    53`
`rx timeout      7       rssi    57`
`rx timeout      9       rssi    60`

`> statistics()`
`Statistics`
`        Uptime          0 days, 11 hours, 46 minutes, 36 seconds`
`        Memory free     11280 bytes`
`        Rx packets      0`
`        Rx packets OK   0`
`        Rx timeouts     1114`
`        Rx forwarded    0`
`        Tx packets      0`
`        Tx packets sent 0`
`        Ack ratio       99.9% (9888/9892)`

I suspect my node module (RN2483) is sending messages, beacuse on 868 MHz it did (with an another gateway on the other end), just reconfigured it to use the 433MHz band, and enabled only one channel, channel 0 for 433.175 MHz operation.

Do you have an idea, how more intensive logging can be enabled, so maybe I can find the reason of not seeing the packets?
Thanks. Regards, Tamás

mogyoros, 2017-04-08:
Strange... now I set the gw to listen on a single SF, SF7, and I can see the packets... 
Of course tx packet is not good, because the backend is not yet modified to use 433MHz settings for this gateway. So seems that I have a problem with the allSF function...

`rxpk	01bb2f00600194f42f0fb1ab	message	{"rxpk":[{"rssi":-46,"stat":1,"modu":"LORA","rfch":1,"tmst":12829931,"datr":"SF7BW125","lsnr":10,"time":"2017-04-08T07:35:28.735724Z","codr":"4/5","data":"AAgHBgUEAwIBmsQbAAujBACSXfhbSZM=","freq":433.175,"chan":0,"size":23}]}	length	237
txpk	{"txpk":{"imme":false,"tmst":18829931,"freq":869.525,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":17,"data":"IFLGJeDDd0yP9yL1eBp1YTg="}}
txpk_ack	{"txpk_ack":{"error":"NONE"}}
transmitPkt	15937	10838	-1	869525000	192	112	2	64	14	17
`

JaapBraam, 2017-04-08:
Have you changed the antenna on your node for a 433Mhz one? What antenna are you using on your gateway?

On the gateway, there is not much extra logging you can enable, because it already works on the limit timing-wise.

A good test to do is to try listening only to a single SF (CONFIG.GW_SF='SF7'). This mode is much less critical timing wise.

JaapBraam, 2017-04-08:
The packet I received on 433Mhz was on a gateway in 'ALL' SF mode. So it can work...

JaapBraam, 2017-04-08:
Or you can make your gateway less sensitive by increasing the RSSI threshold used for signal detection.

https://github.com/JaapBraam/LoRaWanGateway/blob/master/src/SX1276.lua#L366

This will probably decrease the range of your gateway, but will also make it more tolerant for interfering signals.

mogyoros, 2017-04-08:
Well the antenna is the one shipped with the module. This is a non well known and used, chinese [module](https://www.aliexpress.com/item/SX1276-SX3km-1278-433MHz-Wireless-Transceiver-Module-Long-Distance-Receiver-and-transmitter-LoRa-SPSP-Module/32711941138.html). Bought it cause it has SMA connector on the module and they shipped with antenna for a bit less than 10 USD.

For the first trial I will make expriments on single SF, since that seems to work for now. After modifying the backend to use in the downstream RX1 windows I could successfully join. :-)

`rxpk	01b04000600194f42f0fb1ab	message	{"rxpk":[{"rssi":-46,"stat":1,"modu":"LORA","rfch":1,"tmst":1151886069,"datr":"SF7BW125","lsnr":9,"time":"2017-04-08T09:06:02.757319Z","codr":"4/5","data":"AAgHBgUEAwIBmsQbAAujBACbRCCae70=","freq":433.175,"chan":1,"size":23}]}	length	238
txpk	{"txpk":{"imme":false,"tmst":1156886069,"freq":433.175,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":17,"data":"IDUBkmN0mRNCjT/y9f289p4="}}
txpk_ack	{"txpk_ack":{"error":"NONE"}}`

And sending packets also works solid rock.

So you're saying that when in allSF mode, the receiver is configured to listen to **any** kind of radio modulation, not only LoRa modulation on the configured frequency? 
That would explain why I saw with the other single channel gateway implementation from [things4u](https://github.com/things4u/ESP-1ch-Gateway-v4.0) also - which I am also looking at at the moment -  lots of detects and then timeouts are in 868.1MHz...

Can't we configure the LoRa module to listen to only LoRa modulated packets and trigger to the DIO handlers?

mogyoros, 2017-04-08:
One more thing. I mentioned that I am using an ESP12E, but this is a plain ESP8266 board, not an nodemcu built development board. Maybe this is causing issues deep in the nodemcu firmware?
Seems to me something related to the DIO1, since in single SF mode only DIO0 is used, but in allSF DIO0&DIO1...

JaapBraam, 2017-04-08:
In allSF mode, it uses RSSI detection to decide if trying a higher SF is useful. If there are other signals above this threshold present it will make the gateway try the higher SF's increasing the chance to miss the preamble of a lora message... When the preamble is missed, it usually results in a rx timeout...

JaapBraam, 2017-04-08:
I've worked with 'bare' esp8266's too, and have never experienced differences from the Wemos d1 mini's I'm using now...

mogyoros, 2017-04-11:
Indeed, following your suggestion to increase sensitivity, as advised, I have modified the SX1276 lua script to a lower RSSI (38). Now in allSF mode the gateway is able to catch the LoRa modulation radio signals, and gateway forwards packet.
Thanks for your kind help!
(actually currently I am using on 867.3 MHz with the SX1278 module with 433MHz antenna connected, and still it works...)

closed 2017-04-11 by mogyoros

MehrdadBabazadeh, 2018-02-05:
Hi. I also have a problem with 433 MHz using my SX1276MB1MAS. The maximum node-to-node communication range that I can reach for 868, 915 and 433 MHz are 500 m, 500 m, and 20 m respectively. 
I see that the part of the library of sx1272 for 433 MHz includes just four following channels:
case CH_00_433:
case CH_01_433:
case CH_02_433:
case CH_03_433:
It looks like that the library doesn't cover all options for 433 that might be the reason for the short-range issue. 
Is there anyone who has the improved library to use 433 MHz by the SX1276?
Any help is really appreciated