JaapBraam/LoRaWanGateway #28: add localization to the time stamps

dlarue, 2017-04-17:
By adding CONFIG["GW_TZOFFSET"] = "-8" for PST and crate a localtime function instead of gmtime using this offset times the 3600(num sec in 1 hour).  Currently get error though.

local function localtime(t,us)
  -- HOW TO DO THIS?  rtctime.epoch2cal doesn't like my "fixed" epoch time
  local lt = t + CONFIG["GW_TZOFFSET"] * 3600
  local tm = rtctime.epoch2cal(lt)
  return string.format('%04d-%02d-%02d %02d:%02d:%02d GMT',tm["year"],tm["mon"],tm["day"],tm["hour"],tm["min"],tm["sec"])
end

JaapBraam, 2017-04-17:
LoraWAN (at least TTN) expects the dates in the json messages to be GMT. In the TTN console they are displayed in your own timezone (at least, for me that is true).

Why do you want to use localtime?

dlarue, 2017-04-17:
For use in local server and dbases.

dlarue, 2017-04-18:
FYI, my thought was to add it like this so it could be 0 for normal
operation.

CONFIG["GW_TZOFFSET"]="0"
-- PST CONFIG["GW_TZOFFSET"]="-8"