JaapBraam/LoRaWanGateway #32: Output of telnet strange format

Rubensen, 2017-05-07:
Accessing telnet via putty or mobaxterm gives an output unformatted (seems that the lines are too long or without line end). See picture attached.

[image]

It works, buts it´s difficult to read.

JaapBraam, 2017-05-08:
[image]
I think it has to do with your Putty settings. The settings I use work fine...

closed 2017-05-08 by JaapBraam

Rubensen, 2017-05-08:
The Implicit CR in every LF was not checked in mine Putty settings, and I just installed it.
It solves the problem. Thanks.