JaapBraam/LoRaWanGateway #33: hardware for this project

tostmann, 2017-05-24:
Hi!

Its just to tell you that i have inspired by yr project designed a USB-stick-like-hardware to run it on. 
http://busware.de/tiki-index.php?page=LUL

Thanks for yr great work and regards!

[image]

JaapBraam, 2017-05-24:
Nice!

JaapBraam, 2017-05-24:
Are you aware of the fact that 'LUL' means dick in Holland?

closed 2017-05-24 by JaapBraam

tostmann, 2017-05-25:
Ups, didn't know that. Think we might need a new name ;-)

> Am 24.05.2017 um 17:07 schrieb Jaap Braam <notifications@github.com>:
> 
> Are you aware of the fact that 'LUL' means dick in Holland?
> 
>

tostmann, 2017-05-26:
I have renamed my hardware to LUW - Lora-USB-WLAN …
http://busware.de/tiki-index.php?page=LUW

Regards!

JaapBraam, 2017-05-26:
Better!