JaapBraam/LoRaWanGateway #34: Compatibility RF95

DavidBoelee, 2017-06-01:
Hi,

I installed your gateway and it makes contact with TTN as it should, it doesn't receive Lora messages. I am using a RF 95 module. I have been told it should be compatible with the SX1276, but I am wondering whether that's actually true?

I get this message after starting:

```
> got ip	192.168.43.93	255.255.255.0	192.168.43.1
Gateway ID	5CCF7FF42F82E338
ntp synced using 84.80.85.74
2017-06-01 09:44:14 GMT
Unknown radio: 	9
start allSF detector
router ip:	52.169.76.203
ntp synced using 91.148.192.49
```
So the radio is 'unknown'...

Should I get a 'real' SX1276 instead of the RF 95?

JaapBraam, 2017-06-01:
Do you get 'Unknown radio: 9' each time you restart your gateway or does it vary?

If you always get 'Unknown radio: 9' the RF95 module it probably does not contain a SX1276. (I use RFM95 modules)

If the reported radio varies you should check your SPI connections.

DavidBoelee, 2017-06-01:
Tried a couple of times and it's always Unkown Radio 9. I paid a closer look and see that the module I am using is actually a RF 96. It was recognized by the Raspberri Pi gateway I previously used as an SX1272/126 module. Hence the mistake.

I have a couple of other radio modules that I collected over time and see if I can get it wo work with one of those.

Rubensen, 2017-06-02:
Hi

Mine is also using this chip, RF96, and is working and connected to TTN.

Rubens

DavidBoelee, 2017-06-02:
Ok, I wired it up with another RF96 module that had in my drawer and now it's working fine. Couldn't resist trying it with a SX1272 that I had as well, but that's not working  (as expected).

Thanks for your help!

DavidBoelee, 2017-06-02:
I must add that it's working with ABP, I can't get OTAA to work yet. I dived into the other issues regarding OTAA and was wondering whether there is some more documentation available about the settings of the gateway that I can tweak to make it work?

Rubensen, 2017-06-02:
For me only ABP works too.
My gateway is working more than a month right now, and can been seen at
ttnmapper.org Denmark, Ballerup. All mesurements made with fixed SF7.

Nice work JaapBraam! Thanks.

Rubens

JaapBraam, 2017-06-02:
I had some troubles with OTAA too: it was difficult to have a LMIC based node join successfully. It turned out that by specifying a MAX_CLOCK_ERROR of 5% on the node it joined quite reliable.

closed 2017-06-03 by JaapBraam

Rubensen, 2017-06-03:
I´m waiting 10 pieces of RFM95W (aka RF96) to test OTAA again, with the
max_clock_error change. I think this cheap ardunish pro mini modules uses
ressonators instead of crystals, witch could be the cause of the problem.
As I receive 10 prominis today, I will create delays of 100ms and 1s and
verify with a scope how precise they are.

Thanks for the suggetion JaapBraam.

Rubens