JaapBraam/LoRaWanGateway #35: Add status messages

DavidBoelee, 2017-06-02:
Some 'cosmetic' changes:

Maybe in a next version you can add a few extra status messages. When I first started, it seemed it wasn't responding, but then I realised it just couldn't find the wifi network. A message telling that the device is trying to connect to the network would be nice.

Same for the radio module: the software does tell you when an unknown radio is found, but not when a known radio is found. Same confusion: I thought the module had freezed, but it was actually working like a charm.

JaapBraam, 2017-06-02:
Feel free to add those status messages and make a pull request ;-)

DavidBoelee, 2017-06-04:
Good point. :) I will try to when I have more time.