JaapBraam/LoRaWanGateway #36: connect to ttn-router-eu

JBSchueler, 2017-06-09:
Is it possible to connect the gateway to ttn-router-eu?
I would like to[ register my gateway to ttn
](https://console.thethingsnetwork.org/gateways/register)

Can I assume
LoRa Gateway | Register Name | 
---|---|---
GW_HOSTNAME | Gateway ID |
GW_ROUTER | ttn-router-eu |
 | | Description | LoRaWanGateway SF12 and SX1276
??? |  | Gateway Key

Where can I set the provided Gateway Key?

[image]

DavidBoelee, 2017-06-09:
You registered the gateway as a "normal" gateway.  You can use the "I'm using the legacy packet forwarder" checkbox instead. You may have to delete the gateway first and re-register it before you can check it.

I did it this way and everything worked fine. No need to use the gateway key.

closed 2017-06-09 by JaapBraam

JBSchueler, 2017-06-10:
Thanks! It's working well...
When the application starts the Gateway ID is given via the terminal.
I used this 8 digit hex number for the gateway registration.

Now I am receiving data on the dashboard of [https://console.thethingsnetwork.org](https://console.thethingsnetwork.org)
[image]