JaapBraam/LoRaWanGateway #37: PANIC: unprotected error in call to Lua API (bad argument #1 to 'mode' (number expected, got nil))

BertHav, 2017-06-13:
while loading lua files I got some warnings see attached zip.
After several restarts I got the following message.  
PANIC: unprotected error in call to Lua API (bad argument #1 to 'mode' (number expected, got nil))

The config.lua contains:
CONFIG["GW_NSS"]=0
CONFIG["GW_DIO0"]=1
CONFIG["GW_DIO1"]=2
the wiring: NSS to D0, DIO0 to D1 and DIO1 to D2. The modules LolinV3 NodeMCU with ESP-12E and a RFM95W.
[nodemculualog.zip](https://github.com/JaapBraam/LoRaWanGateway/files/1072669/nodemculualog.zip)

Any idea or hint?
Regards Bert

JaapBraam, 2017-06-13:
Can you run CONFIG.print() in order to check your current configuration?

The mode function is used for pin modes. If one of the pin settings is missing in your configuration this may cause the unprotected call...

The values in config.lua are only used if config.json does not exist. Maybe your config.json is not correct. Can you remove it?

Rubensen, 2017-06-14:
It´s always a good idea to erase all memory of the ESP board before upload
a new firmware. Not doing this, could result in some strange errors like
this one.

BertHav, 2017-06-14:
Thanks for coming back to my problem. I have erased the flash. Uploaded the firmware again and the four src files. A config.json does not exist.
The files present:
init.lua          1668 bytes
LoRaWanGW.lc      6928 bytes
Config.lc         2640 bytes
SX1276.lc         8056 bytes

```

NodeMCU custom build by frightanic.com
	branch: master
	commit: 22e1adc4b06c931797539b986c85e229e5942a5f
	SSL: false
	modules: bit,cjson,encoder,file,gpio,net,node,rtctime,sntp,spi,tmr,uart,wifi
 build 	built on: 2017-04-18 19:36
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
no config found, using default values
> CONFIG.print()
Configuration
	GW_HOSTNAME	"bertslorawangw"
	GW_NTP_SERVER	"nl.pool.ntp.org"
	GW_ROUTER  	"router.eu.thethings.network"
	GW_PORT    	"1700"
	GW_FREQ    	868100000
	GW_BW      	"BW125"
	GW_SF      	"ALL"
	GW_LAT     	"52.02"
	GW_LON     	"5.16"
	GW_ALT     	1
	GW_NSS     	0
	GW_DIO0    	1
	GW_DIO1    	2
> got ip	192.168.1.106	255.255.255.0	192.168.1.254
Gateway ID	18FE34F42FF872D1
ntp synced using 95.46.198.21
2017-06-14 20:18:42 GMT
PANIC: unprotected error in call to Lua API (bad argument #1 to 'mode' (number expected, got nil))

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

JaapBraam, 2017-06-14:
What ESP8266 hardware are you using and how much flash memory does it have onboard? I am using WeMos D1 Mini's which have 4Mb flash memory.

The ESP8266Flasher.exe in this repository is configured for ESP8266's with 4Mb flash memory. It flashes two files: the lua firmware and some espressif provided configuration file (esp_init_data_default.bin). The latter has to be flashed on the correct address depending on the flash memory size...

If you have a different flash size look [here](http://nodemcu.readthedocs.io/en/dev/en/flash/) for the correct address. The address can be configured on the config tab of ESP8266Flasher.

BertHav, 2017-06-15:
Jaap, thanks for coming back and pointing in the right direction. I have to blame myself :(. Main problem forget to upload the "esp_init_data_default.bin" and using the wrong lua loader.
Solution: first erase flash, uploading the two bin files, according to config\config.xml 
BTW ESP-12E is also 4MB, according to the reference page, a WeMos D1 Mini is also 4MB and not 4Mb, So the config is unmodified appliccable.
Switched from Windows Lualoader 0.91 to nodemcu-uploader on Linux, uploading the files about 100 times faster and without complaining :).
Now its operational and OTAA works!
Thanks for your tremendous effort!
Gtx Bert

`rx timeout	7	rssi	47
rxpk	0193e20018fe3xxxxxxxxxx	message	{"rxpk":[{"rssi":-75,"stat":1,"modu":"LORA","rfch":1,"tmst":673399837,"datr":"SF9BW125","lsnr":12,"time":"2017-06-15T17:21:33.622149Z","codr":"4/5","data":"AIhVAPB+XXXXXXXXXXXXXXXXXXXXXX=","freq":868.100,"chan":0,"size":23}]}	length	238
txpk	{"txpk":{"imme":false,"tmst":678399837,"freq":868.1,"rfch":0,"powe":14,"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":33,"data":"IF+Zqq9+XXXXXXXXXXXXXXXXXXXXXXXXXX"}}
txpk_ack	{"txpk_ack":{"error":"NONE"}}
rx timeout	8	rssi	47
transmitPkt	16658	12565	-2	868100000	144	112	2	64	14	33
`

closed 2017-06-15 by BertHav