JaapBraam/LoRaWanGateway #39: ESP8266 --> GPIO 13 busy

p1337, 2017-07-15:
Hi there,

I can´t connect the pins of ESP8266 to the SX1278 as you described.

on my Ai-thinker esp8266 12-e      GPIO 13 is connected to GND, otherwise it will not boot up.
where in the Lua files i can change the GPIO 13 to GPIO02 ?

thanks for reply
best regards from Lorch / germany

Robert

JaapBraam, 2017-07-18:
You can use other pins for DIO0 and DIO1, just configure other pins using for instance CONFIG.GW_DIO0=8, followed bij CONFIG.save() to use pin 8 for DIO0. 

You cannot use pin 0 for DIO's because that cannot trigger interrupts.

Rubensen, 2017-07-18:
Hi,

This is just to inform that today my gateway stops sending data and the red
led in my NODEMCU blinks once a second. After some tests I discovered that
one of my three nodes reboots every second due to low voltage of the
battery(it´s a DIY node to test a new circuit). This seems to keep the ESP
to busy to receive any other packet from the others nodes. I will fix this
problem with my node, but I´m afraid that if this happens with a node that
is not mine, that could turns into a big problem.

Rubens

Rubensen, 2017-07-18:
hummm I just saw that TTN was down between 16:11 and 18:33 GMT. Now I could
not confirm that the node was guilty for the fail. BUT, what is fact is
that I could not access via telnet my gateway before a turn off my
defective node and restart my gateway. Only restarting the gateway does not
solve the problem.

On Tue, Jul 18, 2017 at 7:06 PM, Rubens Luciano <rubens.luciano@gmail.com>
wrote:

> Hi,
>
> This is just to inform that today my gateway stops sending data and the
> red led in my NODEMCU blinks once a second. After some tests I discovered
> that one of my three nodes reboots every second [...]

JaapBraam, 2017-07-19:
@Rubensen : This is another issue... Next time (or if the problem persists) please create a new issue...

Anyway:

I think the gateway restarts when status updates to the router fail. Once the status updates to the router succeed again all should be normal again.

This behaviour was not intentionally programmed, but it can be considered a fail-safe feature ;-)

closed 2017-07-19 by JaapBraam