JaapBraam/LoRaWanGateway #43: Problem uploading lua files

covra, 2018-08-28:
I am trying to build a single channel lora gateway with a Lorago dock v1.1 board(ESP8266 and SX1276) and trought ttn I arrived to this repository.  It seems exactly what I want, but,  I found some problems:

- I am uploading lua files with ESPlorer tool, because I dont know how python works very well... Should have I a problem using this tool?

- I gave 4 "succes uploading" (one for each lua file in src folder), but when I reset the board I get:

"Node custom build by.......       ......no config found, use default values" ... and nothing else...

any idea? thank you in advance!

JaapBraam, 2018-08-29:
Have you done the configuration steps described in the Readme?

covra, 2018-08-29:
The only step I had to change was the uploader software, I used ESPlorer tool instead 

> nodemcu-uploader.py 

  even I could see the 4 files uploaded, but "...no config found, use default values" message repeated.... ;(

Other question: I have been reading the "1 channel gateway" posts and it is said that TTN doesnt support 1cgw's (one channel, SF, downlink messages.. etc) ... this firmware works anyway?  i am getting mad cause I have two nodes for testing and I have not none gateway near. This code is "my last chance"  (I tried other codes and this seems the most complet)

thank you again!

JaapBraam, 2018-08-29:
Have you configured the wifi connection as described in 'Register your wifi network'? When the gateway starts it first connects to your wifi network. It will not show any responses until a succesfull connection is made.

It looks like your gateway cannot establish a connection to your network. Check your wifi credentials...

JaapBraam, 2018-08-29:
It should not be necessary to modify your nodes. Just be aware that this gateway will not be able to receive all messages. When using OTA on your node, it will transmit on one of 3 frequencies, so you will have a chance of 1 in 3 that your gateway will receive the OTA request. Your node will try all three frequencies, so be patient, it will get activated. 
When your gateway receives the OTA request TTN will send an OTA respose to the node using your gateway. Your will be connected to TTN now, and will start sending messages. It typically will use one of 8 frequencies of which only one will be received by your gateway so you will have a chance of 1 in 8 that your gateway receives the message. Again: be patient, some messages will be received.

closed 2018-08-30 by JaapBraam