JaapBraam/LoRaWanGateway #44: Problem with lua shell

covra, 2018-08-30:
Using PuTTy under Windows 7 I have a problem with `CONFIG.print() shows the current configuration`

[image]

I also probe `CONFIG.GW_HOSTNAME="LoRaWan_Dock_v1_1" ` and `CONFIG.GW_NTP_SERVER="fr.pool.ntp.org"`  but I dont really know if the parameter was updated.

any idea about this?

covra, 2018-08-31:
I left windows telnet and now I'm working with `ESPlorer v0.2.0-rc5 by 4refr0nt`  (the only problem is I always have to reset in order to connect):
[image]

closed 2018-08-31 by covra

JaapBraam, 2018-08-31:
See the readme:

CONFIG.print() shows the current configuration.
CONFIG.save() saves the current configuration.

Without save, changes in the configuration will only last until a restart.