JaapBraam/LoRaWanGateway #47: why the percent of receive  success is low when set the GW_SF to ALL

Kingwulin, 2018-12-20:
Hi JaapBraam，I am coming again
When I set parameter GW_SF to a specific value(like SF10), the gateway can receive 100% ok，
But when I set the parameter GW_SF to ALL, the gateway mostly prompts "rx timeout", only about 20% can receive the data.So I want to know what about you reality test and if there are any advise for me to  deal this problem
Great thank to you！！！

JaapBraam, 2018-12-20:
Do you have an operational node yourself? Does your gateway receive all messages the node sends on the frequency your gateway is configured to when you set the SF the node is transmitting on? If so, does it still receive all messages when using GW_SF ALL?

Kingwulin, 2018-12-20:
Yeap，I have my own node,and I set its transmitting freq at 471.1MHz , receiving freq at 501.1MHz,.I find it work well when I use the single SF(SF10).But loss many messages when setting the SF to ALL,and I am sure the transmitting and receiving freq of node and gateway as below all the time:
node tx freq = gate rx freq = 471.1MHz
node rx freq = gate tx freq = 501.1MHz

Kingwulin, 2018-12-21:
Hi Jaap Braam
Why not use the RegIrqFlags1 register to determine whether at timeout of receiving but use the value state

[related code here](https://github.com/JaapBraam/LoRaWanGateway/blob/master/src/SX1276.lua#L324)

Kingwulin, 2018-12-21:
I had saw the discussion here, it seems that I know the  reason why I cant receive every message.any way,thank you for your project
by the way, I think value of rssi is unreliable at all.But I still want to know if there were other ways that can listen all sf

Kingwulin, 2018-12-21:
I had saw the discussion [here](https://www.thethingsnetwork.org/forum/t/single-channel-gateway-part-1/798), it seems that I know the reason why I cant receive every message.any way,thank you for your project
by the way, I think value of rssi is unreliable at all.But I still want to know if there were other ways that can listen all sf

garyquan, 2018-12-24:
Hello, I transplanted this project on sx1278 and esp8266, and now I have encountered this problem.

1. When set to fixed SF, the receiving rate is 100%.

2. The same frequency, when set to ALL mode, the receiver rate is about 1%, almost rxtimeout. In all mode, single receiving mode is changed to continuous receiving mode. Data can be received in ALL mode without rxtimeout, but no data can be received for about one minute, which shows that no interruption can be entered. What's the reason for this? Why is there always a timeout in single mode? Have you ever encountered this problem?

Kingwulin, 2018-12-24:
Hi garyquan,I also tested with sx1278 and esp8266，and the problem I meeting is almost the same as you

When setting at ALL sf，I found the gateway can detects the right SF,and enter the single receive mode,but  almost result is rxtimout,so I am sure that using the CAD mode to detect SF is ok,but why cant receive data

what also make puzzled is that why using the rssi value to determine whether enter  rx mode,I just feel that the rssi value is unreliable

Kingwulin, 2018-12-24:
> Hello, I transplanted this project on sx1278 and esp8266, and now I have encountered this problem.
>
> 1. When set to fixed SF, the receiving rate is 100%.
> 2. The same frequency, when set to ALL mode, [...]

hey hey,I found that you are a Chinese too，let have a told over qq or wechat? that maybe would be more Efficient，although we need more JaapBraam's help

JaapBraam, 2019-01-10:
> Hi Jaap Braam
> Why not use the RegIrqFlags1 register to determine whether at timeout of receiving but use the value state
> 
> [related code here](https://github.com/JaapBraam/LoRaWanGateway/blob/master/src/SX1276.lua#L324)

Reading the register takes more time than checking a variable. Reading a register in LUA takes quite some time. It is important to start receiving as fast as possible so that the message won't be missed...

JaapBraam, 2019-01-10:
> Yeap，I have my own node,and I set its transmitting freq at 471.1MHz , receiving freq at 501.1MHz,.I find it work well when I use the single SF(SF10).But loss many messages when setting the SF to [...]

If I remember correctly using lower frequencies, some registers have to be configured differently. Have a look at the SX1276 datasheet. Currently the registers are configured for higher frequencies...

webgps, 2019-12-21:
> Hello, I transplanted this project on sx1278 and esp8266, and now I have encountered this problem.
>
> 1. When set to fixed SF, the receiving rate is 100%.
> 2. The same frequency, when set to ALL mode, [...]

Can you share your transplantation? :)